tree-sitter/tree-sitter-verilog #25: New open source SystemVerilog compliance suite being created

mithro, 2019-09-03:
It might be of interest to you that the [SymbiFlow project](https://symbiflow.github.io) has started working on an open source System Verilog compliance suite which can be found at https://github.com/SymbiFlow/sv-tests

It's still very early stages, but its goal is to cover the whole of the System Verilog LRM. 

The project is currently trying to produce a "status report" for popular open source tools and your tree sitter parser is included. The resulting output report can be found at https://symbiflow.github.io/sv-tests/.

drom, 2019-09-03:
Thank you for the link. How do I include `tree-sitter-verilog` in to report?

mithro, 2019-09-03:
@drom There is a WIP pull request for it at https://github.com/SymbiFlow/sv-tests/pull/13

drom, 2019-09-23:
Using `sv-tests` now.

closed 2019-09-23 by drom