tree-sitter/tree-sitter-verilog #41: Installation error

smgl9, 2020-03-14:
Hi,

I can not figure out why the installation fails on Windows when it was working two/three moths ago. https://github.com/TerosTechnology/colibri/runs/400677109?check_suite_focus=true
In Linux is working fine. Is this something directly related with tree-sitter-verilog?

Thanks in advance, I really apreciate your work on GitHub.

```
> tree-sitter-verilog@0.18.9 install C:\Users\isma\node_modules\tree-sitter-verilog
> node bin/install.js

build
events.js:288
      throw er; // Unhandled 'error' event
      ^

Error: spawn node-gyp ENOENT
    at Process.ChildProcess._handle.onexit (internal/child_process.js:267:19)
    at onErrorNT (internal/child_process.js:469:16)
    at processTicksAndRejections (internal/process/task_queues.js:84:21)
Emitted 'error' event on ChildProcess instance at:
    at Process.ChildProcess._handle.onexit (internal/child_process.js:273:12)
    at onErrorNT (internal/child_process.js:469:16)
    at processTicksAndRejections (internal/process/task_queues.js:84:21) {
  errno: 'ENOENT',
  code: 'ENOENT',
  syscall: 'spawn node-gyp',
  path: 'node-gyp',
  spawnargs: [ 'configure', 'build' ]
}
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\isma\package.json'
npm WARN isma No description
npm WARN isma No repository field.
npm WARN isma No README data
npm WARN isma No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! tree-sitter-verilog@0.18.9 install: `node bin/install.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the tree-sitter-verilog@0.18.9 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```
[2020-03-14T11_43_03_639Z-debug.log](https://github.com/tree-sitter/tree-sitter-verilog/files/4332663/2020-03-14T11_43_03_639Z-debug.log)

smgl9, 2020-03-14:
Taking a look into the logs, it seems that the problem is in a deprecated library, that's the difference between fail and success: npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142

But in Linux is working anyway.

drom, 2020-03-15:
Thank you for reporting. I will try finding Windows machine to debug the issue.

smgl9, 2020-03-19:
I tried what you suggested, following the installation instructions, configuring the paths, etc.  But there is no success :(

qarlosalberto, 2020-05-09:
Any news? 

Thank you!

qarlosalberto, 2020-05-18:
Maybe it's related: https://github.com/georgewfraser/vscode-tree-sitter/issues/28

PR #50 (2023-07-07): fix windows github actions ci
This PR fixes the windows github actions CI build by removing `bin/install.js` and instead using the build script in `package.json` to invoke tree-sitter and node-gyp directly as is done in other tree-sitter grammars.

This should also fix #41 

I also edited the github actions workflows to run on pull requests, in an attempt to show that this PR works, since the travis CI doesn't test windows, but it seems editing workflows requires maintainer approval. 
On my own fork all github actions workflows were passing before I disabled running the CI on push for non-mast [... cut]
closed 2023-07-07 by drom