tree-sitter/tree-sitter-verilog #42: Visual Studio Code extension

qarlosalberto, 2020-05-17:
Hi,

I'm developing a extension for vscode and I'm using tree-sitter-verilog. I have a lot of problems with the NODE_MODULE_VERSION:

```
Activating extension 'teros-technology.teroshdl' failed: The module '/home/carlos/repo/colibri/node_modules/tree-sitter/build/Release/tree_sitter_runtime_binding.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 72. This version of Node.js requires NODE_MODULE_VERSION 75. Please try re-compiling or re-installing the module (for instance, using `npm rebuild` or `npm install`)..
```

If I want to use tree-sitter-verilog, can my extension only run in a specific version of Visual Studio Code cause of the nodejs version?

drom, 2020-05-18:
What is the Node, Electron version you are running under VSCode?

qarlosalberto, 2020-05-18:
For VSCode: electron 7.2.4 and node 12.8.1

I have see that there is a big problem with VSCode an native modules: https://github.com/microsoft/vscode/issues/658

It could be a solution to generate a .wasm language file for tree-sitter-verilog and to use tree-sitter-web? https://github.com/tree-sitter/tree-sitter/tree/master/lib/binding_web#generate-wasm-language-files  

This extension use it: https://github.com/georgewfraser/vscode-tree-sitter/tree/master/parsers

drom, 2020-05-21:
Yes, WASM looks like a promising solution.