tree-sitter/tree-sitter-verilog #45: Generate Error

jussij, 2020-10-15:
I'm running the following 64 bit tree-sitter executable on Windows 10:

```
    D:\Projects\tree-sitter-verilog-master>tree-sitter --version
    tree-sitter 0.17.1 (5080de496a87fc2b9609af80e3d893a04e2b868c)
```

When trying running the generate command the following error is produced:

```
    D:\Projects\tree-sitter-verilog-master>tree-sitter generate
    thread 'main' panicked at 'no entry found for key', cli\src\generate\render.rs:555:41
    note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

The full back trace is as follows:

```
    D:\Projects\tree-sitter-verilog-master>tree-sitter generate
    thread 'main' panicked at 'no entry found for key', cli\src\generate\render.rs:555:41
    stack backtrace:
       0:     0x7ff6c674d9be - <unknown>
       1:     0x7ff6c676d2ec - <unknown>
       2:     0x7ff6c6744a13 - <unknown>
       3:     0x7ff6c675106b - <unknown>
       4:     0x7ff6c6750cb8 - <unknown>
       5:     0x7ff6c675183f - <unknown>
       6:     0x7ff6c67513bf - <unknown>
       7:     0x7ff6c676bbb0 - <unknown>
       8:     0x7ff6c676b952 - <unknown>
       9:     0x7ff6c65bf5fc - <unknown>
      10:     0x7ff6c65d4b1e - <unknown>
      11:     0x7ff6c64f35cc - <unknown>
      12:     0x7ff6c64f0af7 - <unknown>
      13:     0x7ff6c646510b - <unknown>
      14:     0x7ff6c64621f2 - <unknown>
      15:     0x7ff6c646bca6 - <unknown>
      16:     0x7ff6c6751a36 - <unknown>
      17:     0x7ff6c646bc97 - <unknown>
      18:     0x7ff6c678dcb9 - <unknown>
      19:     0x7ffa09fe6fd4 - BaseThreadInitThunk
      20:     0x7ffa0a7dcec1 - RtlUserThreadStart

```

I know this did generate using an earlier release. From memory I think 16.9 worked fine.

drom, 2020-10-15:
@maxbrunsfeld is it safe to use `tree-sitter` v0.17.1 ? should it be reported into `tree-sitter` or `node-tree-sitter` repo?

jussij, 2021-03-10:
Using the latest tree-sitter-verilog files and the latest tree-sitter 0.19.3 release this panic no longer occurs.

I think you can close this bug report as fixed.

closed 2021-03-25 by drom