tree-sitter/tree-sitter-verilog #52: ABI version mismatch

CodePurble, 2021-07-12:
Treesitter for Verilog is not working. When I ran `:checkhealth`, a series of errors citing an ABI version mismatch appear. I have tried running `:TSUpdate verilog` and it is up to date. 

Excerpt from `:checkhealth` output:
```md
health#nvim_treesitter#check
========================================================================
## Installation
  - OK: `tree-sitter` found  0.20.0 (parser generator, only needed for :TSInstallFromGrammar)
  - OK: `node` found v16.4.2 (only needed for :TSInstallFromGrammar)
  - OK: `git` executable found.
  - OK: `cc` executable found. Selected from { vim.NIL, "cc", "gcc", "clang", "cl" }
  - OK: Neovim was compiled with tree-sitter runtime ABI version 13 (required >=13). Parsers must be compatible with runtime ABI.

## Parser/Features H L F I J
  - bibtex         ✓ . ✓ ✓ . 
  - python         ✓ ✓ ✓ ✓ ✓ 
  - comment        ✓ . . . . 
  - lua            ✓ ✓ ✓ ✓ ✓ 
  - json           ✓ ✓ ✓ ✓ . 
  - cpp            ✓ ✓ ✓ ✓ ✓ 
  - c              ✓ ✓ ✓ ✓ ✓ 
  - verilog        x x x . x 
  - devicetree     ✓ ✓ ✓ ✓ ✓ 
  - dockerfile     ✓ . . . ✓ 
  - rust           ✓ ✓ ✓ ✓ ✓ 

 Legend: H[ighlight], L[ocals], F[olds], I[ndents], In[j]ections
         +) multiple parsers found, only one will be used
         x) errors found in the query, try to run :TSUpdate {lang}

The following errors have been detected:
  - ERROR: verilog(highlights): /usr/share/nvim/runtime/lua/vim/treesitter/language.lua:33: ABI version mismatch for ~/.local/share/nvim/site/parser/verilog.so: supported between 13 and 13, found 12
  - ERROR: verilog(locals): /usr/share/nvim/runtime/lua/vim/treesitter/language.lua:33: ABI version mismatch for ~/.local/share/nvim/site/parser/verilog.so: supported between 13 and 13, found 12
  - ERROR: verilog(folds): /usr/share/nvim/runtime/lua/vim/treesitter/language.lua:33: ABI version mismatch for ~/.local/share/nvim/site/parser/verilog.so: supported between 13 and 13, found 12
  - ERROR: verilog(injections): /usr/share/nvim/runtime/lua/vim/treesitter/language.lua:33: ABI version mismatch for ~/.local/share/nvim/site/parser/verilog.so: supported between 13 and 13, found 12
```
Other details:
- Platform: Arch Linux
- `nvim --version`
  - 
    ```
    NVIM v0.5.0
    Build type: Release
    LuaJIT 2.0.5
    Compilation: /usr/bin/cc -D_FORTIFY_SOURCE=2 -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=1 -DNVIM_TS_HAS_SET_MATCH_LIMIT -O2 -DNDEBUG -Wall -Wextra -pedantic -Wno-unused-parameter -Wstrict-prototypes -std=gnu99 -Wshadow -Wconversion -Wmissing-prototypes -Wimplicit-fallthrough -Wvla -fstack-protector-strong -fno-common -fdiagnostics-color=always -DINCLUDE_GENERATED_DECLARATIONS -D_GNU_SOURCE -DNVIM_MSGPACK_HAS_FLOAT32 -DNVIM_UNIBI_HAS_VAR_FROM -DMIN_LOG_LEVEL=3 -I/build/neovim/src/neovim-0.5.0/build/config -I/build/neovim/src/neovim-0.5.0/src -I/usr/include -I/build/neovim/src/neovim-0.5.0/build/src/nvim/auto -I/build/neovim/src/neovim-0.5.0/build/include
    Compiled by builduser

    Features: +acl +iconv +tui
    See ":help feature-compile"

       system vimrc file: "$VIM/sysinit.vim"
      fall-back for $VIM: "/usr/share/nvim"
    ```

abradd, 2021-07-19:
It looks like you are using tree-sitter v0.20.0, based on the package.json file I think the latest parser.c may have been generated from tree-sitter v0.18.x.

Are you regenerating `parser.c` or building the `.so` file from the `parser.c` provided in the repo?

CodePurble, 2021-07-20:
I think I'm using the one from the repo, because I didn't do anything besides `TSInstall verilog`

abradd, 2021-07-21:
I checked the [https://github.com/nvim-treesitter/nvim-treesitter](neovim tree-sitter repo) and it does look like they are using the `parser.c` file from the repo by default.

They have a section on [https://github.com/nvim-treesitter/nvim-treesitter/tree/master/queries/verilog](adding custom parsers) you could use to point to a local instance of this repo if you are willing to generate the parser yourself.

Might be worth posting an issue on the nvim-treesitter repo as well. I imagine others will face this issue until this repo is updated.

CodePurble, 2021-07-28:
I'll take a look at that, thanks!

theHamsta, 2021-07-28:
Actually, we're using an old version of tree-sitter-verilog that supported ABI 13. If nothing went wrong it should install the revision from the lockfile and our CI actually should not touch verilog. It's possible that someone manually change the verilog revision.

https://github.com/nvim-treesitter/nvim-treesitter/blob/9e147d23566f58f2da6682242ffb89cb0de426b5/.github/workflows/update-parsers-pr.yml#L31

CodePurble, 2021-09-14:
I no longer experience this issue. Thanks for all your help!

closed 2021-09-14 by CodePurble