tree-sitter/tree-sitter-verilog #53: I just built a .so file for Python language (using py-tree-sitter) and now then I am trying to use it, it throws an error -  ValueError: Incompatible Language version 12. Must not be between 9 and 11

lhendre, 2021-08-31:
I cloned this repo for the verilog language and am trying to use it with tree sitter; however I am getting the following error.  

ValueError: Incompatible Language version 12. Must not be between 13 and 13

Tried to both manually change the version number and also go back to an older commit but didnt have much luck.

lhendre, 2021-09-14:
Found a work around by using tree-sitter version 0.2.2.  Trying that for the time being as a work around until I can find how o fix this issue and use the current version.

abradd, 2021-09-15:
Were you regenerating the parser? What version of tree-sitter were you using?

I've used 0.19.5 in the past and I didn't have any issues. I did see some issues with 0.20.0, but never looking into it.

lhendre, 2021-10-11:
I was using the most up to date version, will need to go back and grab the version number

closed 2025-07-10 by drom