tree-sitter/tree-sitter-verilog #59: Hihgliting issue on systemverilog file with tree-sitter-verilog on neovim

simoneruffini, 2023-01-29:
Hi I have a highliting issue on a system-verilog file while using tree-sitter on neovim. 
These are my system-verilog lines of code:
[image]
There is some highliting problem because the blue color is used for all the lines after the `""` empty string on top.
The correct output (by disabling tree-sitter) is the following one:
[image]

I hope that this is not too much out of the scope for this repo but I was suggested to post the issue here because it probably relates to this project.

ljjb97, 2024-02-08:
I took a quick look into fixing this issue as I'm seeing it as well. I wasn't able to get a fix working as when I tried to make it optional to have anything in between a string open and close tree-sitter threw an error saying it won't allow tokens that match an empty string. I believe that is a way to prevent [this issue with treesitter.](https://github.com/tree-sitter/tree-sitter/issues/98)

simoneruffini, 2024-02-08:
> I'm also encountering this issue, I think it is because of the way UVM macros use the ` character at the start. Did you manage to find a workaround for this?

but the `'macro` sintax is a valid system-verilog syntax, not just for uvm