tree-sitter/tree-sitter-verilog #72: package.json lack field tree-sitter

zoomer-k, 2024-04-02:
following this guide
https://dcreager.net/2021/06/getting-started-with-tree-sitter/

I add this repository to src folder , but tree-sitter cant detect tree-sitter-verilog

tree-sitter-python has "tree-sitter" field in package.json .

```
  "tree-sitter": [
    {
      "scope": "source.python",
      "file-types": [
        "py"
      ],
      "injection-regex": "py",
      "highlights": "queries/highlights.scm",
      "tags": "queries/tags.scm"
    }
  ]
```

when add this field , tree-sitter cli can parse

clason, 2024-04-27:
Fixed in the last commit @amaanq

zoomer-k, 2024-11-21:
@clason @amaanq 
dose someone check this merge ?

filetypes field is lack 
injection-regex is not collect 

```
 "tree-sitter": [
    {
      "scope": "source.verilog",
      "file-types":[
        "sv"
      ],
      "injection-regex": "sv"
    }
```

↑ maybe collect