tree-sitter/tree-sitter-verilog #79: tree-sitter.json is lack of filed

zoomer-k, 2024-12-17:
```
  "grammars": [
    {
      "name": "verilog",
      "camelcase": "Verilog",
      "scope": "source.verilog",
      "path": ".",
      "file-types": ["sv"],
      "injection-regex": "sv"
    }
```

my enviroment 
node v18.16.0

tree-sitter 0.24.5 (1a983b7e2c45d9b9f45b2133a4e3a829f7e961be)

drom, 2025-07-10:
@zoomer-k 
Would you mind to explain more about your issue?

zoomer-k, 2025-07-10:
Hi @drom 
package.json was updated in this issue
https://github.com/tree-sitter/tree-sitter-verilog/issues/72

but It moved to tree-sitter.json 
https://github.com/tree-sitter/tree-sitter/pull/3700

https://tree-sitter.github.io/tree-sitter/cli/init.html?highlight=tree-sitter.json#structure-of-tree-sitterjson