tree-sitter/tree-sitter-verilog #80: Why I was removed from tree-sitter org?

drom, 2025-01-05:
Why I was removed from tree-sitter org?
Anybody can explain?
@amaanq @dundargoc @nathansobo @ObserverOfTime @rewinfrey @WillLillis 

I am the original developer of tree-sitter-verilog
I moved tree-sitter-verilog into tree-sitter org to collaborate there.
What was the reason of removal from org?

amaanq, 2025-01-05:
I was cleaning up the org, since there were too many members and outside collaborators that hadn't been active in several years, just to reduce the # of people that can directly modify/commit to any of the repos. Sorry about that, I'll get you added back (but in the future, don't tag a bunch of people....)

drom, 2025-01-06:
I did not know who removed me from the ORG so the list was long.
I still don't see myself in the ORG after accepting the invite.

amaanq, 2025-01-06:
You have the maintainer role in this repo, so you have full permissions to write/edit the repo as you please, but do you need to be in the org? Not to offend you, but I don't see the point in adding you to an org if you just need access to 1 repo

closed 2025-01-06 by amaanq