tree-sitter/tree-sitter-verilog #81: Fails to parse correct code

numero-744, 2025-03-04:
Hi there 👋

I have tried to use [`mergiraf`](https://codeberg.org/mergiraf/mergiraf/releases/tag/v0.6.0) on https://github.com/openhwgroup/cva6/blob/master/core/scoreboard.sv and it fails to parse this file using `tree-sitter-verilog`.
The `scoreboard.sv` file is correctly parsed by verilator, questa and vcs, but not by `tree-sitter-verilog` 🤔
Well, CVA6 looks like an interesting playground for integration testing 😉

There seem to be different errors that I have not yet identified.
(Maybe system tasks (eg: `$clog2`) and for loops?)

I have reproduced the issue running `tree-sitter-cli` directly.

```bash
% tree-sitter parse --scope source.verilog core/scoreboard.sv
...
core/scoreboard.sv	Parse:    3.94 ms	 3893 bytes/ms	(ERROR [121, 4] - [121, 61])
```

Is there anybody motivated to fix this?
Else I can have a look this week-end.

I have the following document:

> SystemVerilog 3.1a
> Language Reference Manual
> Accellera’s Extensions to Verilog®

It seems the standard you have implemented is not the one I have.
The sections have changed so I cannot map between `grammar.js` and the specification.

PS: Wondering if `tree-sitter` performs pre-processing for `` `include `` and macros.
Otherwise, are there workarounds to avoid false errors on ugly but correct code?

zoomer-k, 2025-03-10:
https://github.com/gmlarumbe/tree-sitter-systemverilog

its better than this repository

numero-744, 2025-03-10:
Hello!

Thanks for the link, and thanks for the that is a lot of work that is made there!

However, I'm getting other errors, for the same file:

```
% tree-sitter parse --scope source.verilog core/scoreboard.sv
...
core/scoreboard.sv	Parse:    5.13 ms	 2989 bytes/ms	(ERROR [201, 18] - [201, 20])
tuxedo-come ~/Development/cva6                                         8:53
% tree-sitter langs
scope: source.verilog
parser: "/home/come/Development/tree-sitter/tree-sitter-systemverilog/."
highlights: None
file_types: []
content_regex: None
injection_regex: Some(Regex("^verilog$"))
```

Here is line 202, column 19:

```
            mem_n[8'(trans_id_i[i])-1].sbe.valid = 1'b1;
                   ^ this type casting operation seems to trigger a parse error
```

I'll open issues there. Thanks again!